prueker/METARMap #3: metar.py error

lynolamero, 2020-05-10:
running the new metar.py prints out an error:

Traceback (most recent call last):
  File "metar.py", line 109, in <module>
    pixels[i] = color
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 176, in __setitem__
    self._set_item(index, val)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 131, in _set_item
    raise IndexError
IndexError

And, when running refresh, I get a pkill error, as follows:
pi@METAR-MAP:~ $ ./refresh.sh
pkill: pidfile not valid
Try `pkill --help' for more information.
pi@METAR-MAP:~ $ https://www.aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=5&mostRecentForEachStation=true&stationString=KALW,KPSC,KLWS,KPUW,KGEG,KSFF,KDEW,KCOE,KSZT,K65S,CYCG,CYLW,CYYF,CYDC,CYHE,CWAE,CYPW,CYCD,CYVR,CYXX,KBLI,KORS,KFHR,CYYJ,KUIL,KCLM,K0S9,KBVS,KAWO,KPAE,KBFI,KRNT,KBWT,KSHN,KTIW,KPLU,KOLM,KCLS,KHQM,KAST,K1S5,KYKM,KELN,KEAT,KSMP,KEPH,KOMK,KS52
Missing flight condition, skipping.
Missing Wind speed, skipping.
Setting LED 0 for KALW to VFR (255, 0, 0)
Setting LED 1 for KPSC to VFR (255, 0, 0)
Setting LED 2 for KLWS to VFR (255, 0, 0)
Setting LED 3 for KPUW to VFR (255, 0, 0)
Setting LED 4 for KGEG to VFR (255, 0, 0)
Setting LED 5 for KSFF to VFR (255, 0, 0)
Setting LED 6 for KDEW to VFR (255, 0, 0)
Setting LED 7 for KCOE to VFR (255, 0, 0)
Setting LED 8 for KSZT to VFR (255, 0, 0)
Setting LED 9 for K65S to VFR (255, 0, 0)
Setting LED 10 for CYCG to VFR (255, 0, 0)
Setting LED 11 for CYLW to VFR (255, 0, 0)
Setting LED 12 for CYYF to VFR (255, 0, 0)
Setting LED 13 for CYDC to VFR (255, 0, 0)
Setting LED 14 for CYHE to VFR (255, 0, 0)
Setting LED 15 for CWAE to VFR (255, 0, 0)
Setting LED 16 for CYPW to VFR (255, 0, 0)
Setting LED 17 for CYCD to VFR (255, 0, 0)
Setting LED 18 for CYVR to VFR (255, 0, 0)
Setting LED 19 for CYXX to VFR (255, 0, 0)
Setting LED 20 for KBLI to VFR (255, 0, 0)
Setting LED 21 for KORS to VFR (255, 0, 0)
Setting LED 22 for KFHR to VFR (255, 0, 0)
Setting LED 23 for CYYJ to VFR (255, 0, 0)
Setting LED 24 for KUIL to VFR (255, 0, 0)
Setting LED 25 for KCLM to VFR (255, 0, 0)
Setting LED 26 for K0S9 to None (0, 0, 0)
Setting LED 27 for KBVS to VFR (255, 0, 0)
Setting LED 28 for KAWO to VFR (255, 0, 0)
Setting LED 29 for KPAE to VFR (255, 0, 0)
Setting LED 30 for KBFI to VFR (255, 0, 0)
Setting LED 31 for KRNT to VFR (255, 0, 0)
Setting LED 32 for KBWT to None (0, 0, 0)
Setting LED 33 for KSHN to VFR (255, 0, 0)
Setting LED 34 for KTIW to VFR (255, 0, 0)
Setting LED 35 for KPLU to VFR (255, 0, 0)
Setting LED 36 for KOLM to VFR (255, 0, 0)
Setting LED 37 for KCLS to VFR (255, 0, 0)
Setting LED 38 for KHQM to VFR (255, 0, 0)
Setting LED 39 for KAST to VFR (255, 0, 0)
Setting LED 40 for K1S5 to None (0, 0, 0)
Setting LED 41 for KYKM to VFR (255, 0, 0)
Setting LED 42 for KELN to VFR (255, 0, 0)
Setting LED 43 for KEAT to VFR (255, 0, 0)
Setting LED 44 for KSMP to VFR (255, 0, 0)
Setting LED 45 for KEPH to VFR (255, 0, 0)
Setting LED 46 for KOMK to VFR (255, 0, 0)
Setting LED 47 for KS52 to VFR (255, 0, 0)
Traceback (most recent call last):
  File "/home/pi/metar.py", line 109, in <module>
    pixels[i] = color
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 176, in __setitem__
    self._set_item(index, val)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 131, in _set_item
    raise IndexError
IndexError

prueker, 2020-05-10:
That sounds like you are addressing more LEDs than you initialized.
Make sure to set the LED_COUNT appropriate to the number of LEDs being used.

The pkill error is fine and you can ignore it, it's just a preventative operation for if another version of the same script is already running to make sure it gets closed before starting the next one.

lynolamero, 2020-05-10:
Yes, LED count was off by one due to first LED assigned 0.

closed 2020-05-10 by lynolamero

prueker, 2020-05-10:
Great, glad I could help.