prueker/METARMap #5: NameError: name 'lightningConditions' is not defined

supportsolutionsit, 2020-06-11:
Hi,
Tried your updated scripts on a new pi following your instructions on what packages to use etc  and copied the files over to /home/pi/ (all default except LED_Count = 35 and my airports file - UK Metar sites).
I get the following error;
**Traceback (most recent call last):
  File "metar.py", line 110, in <module>
    print("Setting LED " + str(i) + " for " + airportcode + " to " + ("lightning " if lightningConditions else "") + ("windy " if windy else "") + (conditions["flightCategory"] if conditions != None else "None") + " " + str(color))
NameError: name 'lightningConditions' is not defined**

I can see where in metar.py it falls over;
**print("Setting LED " + str(i) + " for " + airportcode + " to " + ("lightning " if lightningConditions else "") + ("windy " if windy else "") + (conditions["flightCategory"] if conditions != None else "None") + " " + str(color))** and if I remark that line out if falls over on line 111 with;
 **Traceback (most recent call last):
  File "metar.py", line 111, in <module>
    pixels[i] = color
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 298, in __setitem__
    self._set_item(index, r, g, b, w)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 264, in _set_item
    raise IndexError
IndexError**

any idea's ?

Regards
D

prueker, 2020-06-11:
Ok, so I just made a small fix that might be why you had the issue about lightningConditions (basically indenting this like by 1 tab to make sure the variable is in scope).

For the second issue about IndexError - that looks like you are trying to set more LEDs than you initialized. Make sure you set the LED_COUNT to at least the amount of LEDs you are addressing (nulls included).

supportsolutionsit, 2020-06-11:
Hi Philip.
All good with defining Windy = False & lightningCondition = False
Should have spotted that. 
All good to close
Thank you
D

closed 2020-06-11 by prueker