prueker/METARMap #6: Index Error

o62, 2020-06-11:
Hi! I updated the airports list and the metar.py file but get this error at the end:

`Setting LED 53 for KBFL to VFR (255, 0, 0)
Setting LED 60 for KMMH to VFR (255, 0, 0)
Traceback (most recent call last):
  File "metar.py", line 111, in <module>
    pixels[i] = color
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 298, in __setitem__
    self._set_item(index, r, g, b, w)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 264, in _set_item
    raise IndexError
IndexError
`

prueker, 2020-06-11:
An IndexError suggests that you didn't set the LED_COUNT high enough.

Make sure that the LED_COUNT is set to at least the number of LEDs you are using.

supportsolutionsit, 2020-06-11:
Diffidently was the wrong LED count I dont have 60 LED's (setting LED 60 for KMMH) - only 46 . - I typo'd my original Post with 35 but I had 45 in my metar.py.

You can close this error. - Thank you for your help.

closed 2020-06-11 by prueker