prueker/METARMap #8: Python3 can't open file 'metar.py' : [Errno 2]No such file or directory

micahclaussen, 2020-07-03:
Hi! I cloned the repository into my pi with "git clone https://github.com/prueker/METARMap" and when I try to run metar.py, it says that it can't find the file.

I don't think I copied the files correctly. GitHub says "Copy the metar.py, pixelsoff.py, airports, refresh.sh and lightsoff.sh scripts into the pi home directory" but I'm not sure this is what I did.

First time doing any sort of coding, sorry.

prueker, 2020-07-03:
Make sure you're in the right folder.
When you run git clone, it will create a folder with the files, so you need to change into that folder to run the files.

You can type "ls" to see the list of files in the current folder you are in and "cd FOLDERNAME" to change into a folder.

micahclaussen, 2020-07-03:
Thanks so much! I’ll let you know how it goes.

closed 2020-07-03 by prueker