prueker/METARMap #9: Airports file not found

micahclaussen, 2020-07-04:
When running “sudo python3 metar.py” from the METARMap directory, it says ‘/home/pi/airports’ can’t be found...

The airports file is listed when you input “ls” into the METARMap directory along with the other files, “crontab, license, lightsoff.sh” etc...

Is this where the airport list should be? Thanks again for your help.

prueker, 2020-07-04:
So it sounds like you have everthing in a subfolder "METARMap" right now.
The script is set up to run directly in the root user folder (/home/pi).

The easiest thing will be to move all the files into the root:
Go into the METARMap folder you have, then type 
`mv * ../`
This will move all the files in the current folder into the parent folder.

then change into the /home/pi folder using cd command.

micahclaussen, 2020-07-04:
Thanks so much for your help. After 2 weeks of

closed 2020-07-04 by micahclaussen