prueker/METARMap #12: Data not being sent as light output

AVIDPilot, 2020-11-02:
I have used the software for a while but I just rebooted my pi and code no longer works. The code will run through my airports (Michigan Airports) and give out a true false statement....
KMTC:VFR:12:True:False
KJXN:VFR:10:True:False
KBAX:VFR:11:True:False
KCFS:VFR:7:False:False
KMGN:VFR:5:False:False

But code ends with....
Traceback (most recent call last):
  File "metar.py", line 110, in <module>
    print("Setting LED " + str(i) + " for " + airportcode + " to " + ("lightning " if lightningConditions else "") + ("windy " if windy else "") + (conditions["flightCategory"] if conditions != None else "None") + " " + str(color))
NameError: name 'lightningConditions' is not defined

I updated and reinstalled all download files. When I run python3 code for all lights to come on in one color it works (in all the colors). This strictly seems like a code pushing information to the board or to python error. I am not fluent with python so any help would be awesome. 

Full code list of command run....(no lights on)
pi@raspberrypi:~ $ sudo python3 metar.py
https://www.aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=5&mostRecentForEachStation=true&stationString=KADG,,KARB,KYIP,KDTW,,KMTC,,KPTK,,KFNT,KRNP,,KCFS,,KBAX,,,KMBS,,KAMN,KMOP,,,Y31,,KOSC,,KAPN,KPZQ,,KSLH,,KDRM,,KCIU,,KERY,,KISQ,,KSJX,,KMGN,KCVX,KACB,KGLR,KGOV,KHTL,,KCAD,,KTVC,,KFKS,KMBL,KLDA,,KRQB,,KFFX,KMKG,,KBIV,,8D4,KGRR,Y70,KLAN,KTEW,KJXN,KJYM,KOEB,,KBTL,KAZO,,KLWA,,KSBN
KISQ:VFR:4:False:False
KGOV:MVFR:6:False:False
KMBL:VFR:7:False:False
KCIU:MVFR:4:False:False
KMTC:VFR:12:True:False
KJXN:VFR:10:True:False
KBAX:VFR:11:True:False
KCFS:VFR:7:False:False
KMGN:VFR:5:False:False
KERY:VFR:4:False:False
KLWA:VFR:7:False:False
KOEB:VFR:4:False:False
KRQB:VFR:3:False:False
KCAD:MVFR:7:False:False
KJYM:VFR:7:False:False
KRNP:VFR:7:False:False
KFKS:VFR:5:False:False
KOSC:VFR:13:True:False
KPZQ:MVFR:8:False:False
KFFX:VFR:3:False:False
KSJX:VFR:5:False:False
KACB:VFR:4:False:False
KCVX:VFR:8:False:False
KTEW:VFR:6:False:False
KMOP:VFR:6:False:False
KMKG:VFR:5:False:False
KAMN:VFR:6:False:False
KDRM:VFR:5:False:False
KSBN:VFR:6:False:False
KAPN:VFR:12:True:False
KARB:VFR:10:True:False
KYIP:VFR:11:False:False
KBIV:VFR:4:False:False
KPTK:VFR:14:True:False
KGRR:VFR:5:False:False
KTVC:VFR:6:False:False
KBTL:VFR:5:False:False
KAZO:VFR:6:False:False
KLAN:VFR:8:False:False
KDTW:VFR:17:True:False
KFNT:VFR:12:True:False
KMBS:VFR:9:False:False
Missing flight condition, skipping.
KHTL:VFR:11:True:False
KGLR:VFR:5:True:False
KSLH:VFR:11:True:False
Traceback (most recent call last):
  File "metar.py", line 110, in <module>
    print("Setting LED " + str(i) + " for " + airportcode + " to " + ("lightning " if lightningConditions else "") + ("windy " if windy else "") + (conditions["flightCategory"] if conditions != None else "None") + " " + str(color))
NameError: name 'lightningConditions' is not defined

prueker, 2020-11-02:
Can you make sure you have the latest version of the metar.py file? https://github.com/prueker/METARMap/blob/master/metar.py
The error indicates that the lightningConditions variable isn't defined.
The last change I made specifically defines it in line 97 - https://github.com/prueker/METARMap/blob/2e41aedc7babfaed6e5eb733cd8dacf48bfe8aaa/metar.py#L97

AVIDPilot, 2020-11-03:
That seemed to do the trick. strange that it worked for so long without the update. Thank you so much for making the code and also being awesome at keeping it up to date.

closed 2020-11-03 by AVIDPilot