prueker/METARMap #14: Refresh.sh not consistently acting

broort22, 2020-12-08:
Hi,

Please bear with me as I am very inexperienced with Linux and Raspberry Pi.  I'm amazed I've gotten as far as I have.  It wasnt until today I realized you updated the metar script to factor in wind.  I've updated my metar and refresh script with what you currently have published however periodically when it refreshes I'll notice that it stops factoring wind when it should be there.  When i run the metar script manually it seems to put it back on track.

Do you think you could give me any pointers on where to begin to try and diagnose this issue?

Thanks so much for the fantastic work you have put into this.  I have significantly enjoyed analyzing my own built Metar Map

prueker, 2020-12-08:
Did you update the refresh.sh file to the latest from here?
One thing I added for the winds is that the script runs for 5 minutes and then the refresh.sh script kicks off a new round and kills the previous instance (to make sure there's no collisions using the temporary metarpid.pid file where it stores the latest script run id, so if you do have the latest refresh.sh file, I'd suggest you check whether the metarpid.pid file was created correctly and is accessible.

broort22, 2020-12-08:
It’s hard to say what the problem is to troubleshoot when after I woke up this morning it was working fine for the full hour before I left for work.  This evening when I return I’ll verify if it still looks in order.  Thanks for reaching out so quick!

prueker, 2020-12-12:
Since you haven't replied back, I'm guessing it's working now?

closed 2020-12-12 by prueker

broort22, 2020-12-12:
Yes Phil, it’s working. I’m not certain what I did if anything to help it consistently cycle but it works tremendously.  

Appreciate all your work and follow up to help out with this.