prueker/METARMap #15: Dimming at night instead of off

nesincg, 2020-12-12:
Have you thought about changing code to dim the LEDs during the evening hours. Find local sunset and sunrise times
https://stackoverflow.com/questions/38986527/sunrise-and-sunset-time-in-python/38986561

I know just about nothing about programming. Only a tiny bit from crappy javascript skills, but I may give it a shot. 

You could also keep the shutoff times too. My LEDs are stupid bright and need to be dimmed to .4 during day and .1 at night. Full brightness lights up the hangar.

Thanks for everything, this has been a fun learning experience.

proppilot, 2020-12-12:
I would be willing to share the code I developed that uses a map mounted motion sensor to trigger the lights on for 30 minutes. The timer resets with motion so they darken 30 minutes after no motion.  Off when the room is empty and dramatic improvement to the lifespan of the LEDS.

Regards,

Tony Varga

nesincg, 2020-12-13:
Tony, I don't really know how to use Github. I would still like to dim at night, but your feature sounds awesome. Can you send me or Philip that code? Did you use a USB motion sensor?

prueker, 2020-12-13:
Hey,

Thanks for the suggestion. I've implemented it now, based on either a fixed time of day to start dimming, or sunrise/sunset of nearby city.
You can find the instructions to set it up in the readme: https://github.com/prueker/METARMap#additional-led-dimming-functionality-based-on-time-of-day

closed 2020-12-13 by prueker