prueker/METARMap #16: Astra error.

lynolamero, 2020-12-13:
After running: sudo pip3 install astral
I got: Successfully installed astral-2.2 pytz-2020.4

I updated: metar.py with new code, then ran metar.py and got this:

Traceback (most recent call last):
  File "metar.py", line 67, in <module>
    ast = astral.Astral()
AttributeError: module 'astral' has no attribute 'Astral'

As coded:
# Figure out sunrise/sunset times if astral is being used if astral is not None and USE_SUNRISE_SUNSET:
	ast = astral.Astral()
	try:
		city = ast[LOCATION]
	except KeyError:
		print("Location not recognized, please check list of supported cities and reconfigure")
	else:
		print(city)
		sun = city.sun(date = datetime.datetime.now().date(), local = True)
		BRIGHT_TIME_START = sun['sunrise'].time()
		DIM_TIME_START = sun['sunset'].time()
		print("Sunrise:" + BRIGHT_TIME_START.strftime('%H:%M') + " Sunset:" + DIM_TIME_START.strftime('%H:%M'))

# ----- Daytime dimming of LEDs based on time of day or Sunset/Sunrise -----
ACTIVATE_DAYTIME_DIMMING = True			# Set to True if you want to dim the map after a certain time of day
LED_BRIGHTNESS_DIM	= 0.1				# Float from 0.0 (min) to 1.0 (max)

BRIGHT_TIME_START = datetime.time(7,0)			# Time of day to run at LED_BRIGHTNESS in hours and minutes
DIM_TIME_START =	datetime.time(21,0)		# Time of day to run at LED_BRIGHTNESS_DIM in hours and minutes

USE_SUNRISE_SUNSET = True				# Set to True if instead of fixed times for bright/dimming, you want to use local sunrise/sunset
LOCATION = "Seattle"					# Nearby city for Sunset/Sunrise timing, refer to https://astral.readthedocs.io/en/latest/#cities for list of cities supported

Why the error?

prueker, 2020-12-13:
just to check, how did you run metar.py?
Using the refresh.sh script, or by running python directly?
Make sure you're running python3, since that's what pip3 installs it for.

You can test that Astral installed correctly by entering python3 in the console if you run "python3" and then type
`import astral
astral.`
and then hit the tab key twice, which brings up all the objects and in there you should see a astral.Astral()
[image]

prueker, 2020-12-13:
Nevermind my earlier comment, I see the issue now. You are running a newer version of python, which is using a newer version of Astral than I was using.
I haven't re-imaged my Raspberry in a while, so I am running python 3.5, which is using Astral 1.10.1 and the newer version 2.2 of Astral has some breaking changes.

I'm updating my Raspberry now and will make a fix that will be compatible with both old and new version (since most people that built their maps a while ago will likely still run python 3.5 like I did.

prueker, 2020-12-13:
I just checked in the fix, thanks for the good catch.

The new version of metar.py is compatible with both older python 3.5 as well as newer python 3.6+ clients.

closed 2020-12-13 by prueker

lynolamero, 2020-12-14:
Thanks for the update. I appreciate your efforts and talent. 
I’ve installed the new metar.py file. I’ll see if it dims tonight. 

Regards,
Lynol