prueker/METARMap #24: Displayairports

lynolamero, 2021-05-08:
I've added the displayairports file to the root directory with a truncated list of airports I want to show on the mini-display.  However, I got this error when running metar.py:
.
.
.
Setting LED 47 for KS52 to VFR (255, 0, 0)
Traceback (most recent call last):
  File "metar.py", line 248, in <module>
    displaymetar.outputMetar(disp, stationList[displayAirportCounter], conditionDict.get(stationList[displayAirportCounter], None))
IndexError: list index out of range

Does 1.4.3 not include code for the truncated displayairports list?

prueker, 2021-05-08:
Make sure you don't have a blank line at the end of your displayairports file.
It should be just the airports, so something like this for example.
[image]

lynolamero, 2021-05-08:
Verified and changed to plain text document.  working.

prueker, 2021-05-08:
Great, glad I could help figure it out.

closed 2021-05-08 by prueker

lynolamero, 2021-07-03:
Hi Philip,
The metar map has not be displaying for the last few days.  I’ve restarted and still nothing, so I connected and ran sudo python3 metar.py and got this traceback errors:

***@***.***:3SM:BR:13/13:30.03:False
***@***.***:10SM::31/10:29.86:False
***@***.***:8SM::16/14:30.04:False
***@***.***:10SM::27/12:29.94:False
setting up external display
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 154, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 87, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 38, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 256, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 160, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 77, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 44, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end-start)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 155, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "metar.py", line 199, in <module>
    disp = displaymetar.startDisplay()
  File "/home/pi/displaymetar.py", line 18, in startDisplay
    disp = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ssd1306.py", line 220, in __init__
    self.i2c_device = i2c_device.I2CDevice(i2c, addr)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 163, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x3c

Can you help me figure out what’s happening?

Best Regards,
Lynol

prueker, 2021-07-03:
@lynolamero the error sounds like the Raspberry Pi is not detecting the external display, maybe the wire connection got lose or the display failed.

lynolamero, 2021-07-06:
Hi Philip,
I’ve checked all the connections.  They are secure.
As part of the troubleshooting, I set the external display to “False” and the LEDs all work fine. So, yes, the external display is at fault. I may need to just replace the external display.  Funny it failed so soon. 
Do you have any other ideas?

Best Regards,
Lynol

prueker, 2021-07-07:
You could rerun the commands I have on the instructions to see if the raspberry pi even still detects it, but other than that I have no useful other troubleshooting tip and the error message does hint at connection issue/faulty unit.