prueker/METARMap #26: Board 'hangs' with no status updates (Out of Memory)

n2jwc, 2021-09-07:
FIgured this was the best way to let you know of an issue I have seen twice.  about 7 days of running 24/7 and eventually the board never updates.  With a clean boot and all working, running refresh.sh works perfectly. However when it is 'hung' and I attempt the same shell script, I see the following and only a reboot fixes it.

```
Traceback (most recent call last):
  File "/home/pi/metar.py", line 272, in <module>
    pixels.show()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pixelbuf.py", line 197, in show
    return self._transmit(self._post_brightness_buffer)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 164, in _transmit
    neopixel_write(self.pin, buffer)
  File "/usr/local/lib/python3.7/dist-packages/neopixel_write.py", line 34, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 80, in neopixel_write
    "ws2811_init failed with code {0} ({1})".format(resp, message)
RuntimeError: ws2811_init failed with code -2 (Out of memory)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```

bleeblak8, 2022-01-15:
Hi n2jwc, 
I also noticed that my map was freezing after a while and found the same errors that you found.  From doing some investigating, it appears to be an issue with the neopixel library.  I'm investigating on if there is a change that can be made.  Once I have found a solution for my setup, I will update this post.
https://github.com/jgarff/rpi_ws281x/issues/390
Thanks

cbronner35, 2022-12-10:
Solution for me is when I created a cronjob I setup a reboot once a day. Basically solved all my problems just an fyi

prueker, 2022-12-10:
Yes, it seems an inherent problem with the underlying neopixels and limited memory of the Raspberry Pi over time, but rebooting fixes it, so closing the issue.

closed 2022-12-10 by prueker