prueker/METARMap #33: Reporting sites that aren't pulling either due to be down or under maintenance make the LED turn off

cbronner35, 2022-02-04:
Just as the subject line said. Was wondering if it were possible to light up the LED orange if for some reason it has trouble fetching from a particular reporting side. People always think I have an LED light out when one reporting site fails lol

prueker, 2022-02-04:
Yes, you can do this easily,just define another color at the top, so if you wanted orange, you could define
`COLOR_ORANGE	= (255,128,0) 		# Orange`
and then replace the default for when there's no condition down in the loop **around line 233** from
`		color = COLOR_CLEAR`
to your newly defined color like
`		color = COLOR_ORANGE`

Then it will show up in orange if there's no condition reported currently.

closed 2022-02-04 by prueker

cbronner35, 2022-02-04:
I appreciate that should work good!!