prueker/METARMap #39: Modify for Raspberry Pi Pico?

derekantrican, 2022-11-15:
Is there a chance this project could be modified to work for the Raspberry Pi Pico? I don't really see why not (unless there's some missing feature of micro python that I'm not aware of). Mostly seems like a few concepts based on a "regular" raspberry pi & a linux system would have to be changed.

prueker, 2022-12-10:
Yes, it might work assuming that the micro python libraries can support the neopixel libraries and that memory for loading the live weather data doesn't run into issues.

I'm a little busy at the moment, so I don't have enough time to tinker around to try it, but if someone wants to go try, feel free to.

tiuchs, 2023-07-04:
@derekantrican did you succeed in porting the project to a Raspberry Pi Pico?

derekantrican, 2023-07-04:
No, I haven't put any effort into this. I was just asking out of theory

closed 2023-10-18 by prueker