prueker/METARMap #46: Change the intensity of specific LEDs

speedtoys, 2023-09-19:
I have a 300 LED project, that is already setup, epoxied, and..done..

But one string in the middle is maybe 20% brighter than the rest..FML...

Can you introduce an update that will let you define chunks of led brightness as the base for daytime/nighttime settings

0-199 .8
200-249 .6
250-299 .8

TY

cbronner35, 2023-09-19:
Did you not test the Leds before epoxying? I'm not sure about how to code it but if the strips are powered separately then I'd use a resistor to dim the middle one. But there are other electrical options if your injecting power on both sides too

speedtoys, 2023-09-19:
They "looked" the same, but in the board, with the 'lenses' installed (4x12mm washers in the holes to narrow down the individual LED footprint) map down, epoxy on...this ONE string is markedly brighter.

I do have wiring access..yes.

cbronner35, 2023-09-19:
Oh I gottcha yeah if someone can't find a code solution I'd just test a few resistors and see if I can dim it a little bit to match the others

speedtoys, 2023-10-01:
You wont believe this..

I swapped out that last string on the chain to match the strings ahead of them.

_They're still a lot brighter_ than the rest of the LEDs.

I did some additional 5v distribution, and now every LED is within .2v of any other, so its not voltage either.

You can see in the image that the SW US, AK, HI, and the Legend are wayyy brighter than the 250 LEDs before them, only LEDs 250 on up, are brighter.

[image]
[image]
[image]
[image]

It's gotta be in the code..its not chain make or model, or voltage.

speedtoys, 2023-10-03:
When it dims the LEDs for night mode, the last string does not diminish in intensity either.  The first 250 do.

speedtoys, 2023-10-08:
This pattern follows at any LED brightness setting, have now tried a THIRD same vendor string at address 249-300.

This last string is always brighter, relative to the "LED_BRIGHTNESS" value, than the 250 LEDs before it.

prueker, 2023-10-18:
To troubleshoot the issue further, I would try to run just the LEDs by themselves, without the script as I mention at https://slingtsi.rueker.com/making-a-led-powered-metar-map-for-your-wall/

The easiest way to test this once you’ve installed the software is to just manually control the colors of the LEDs by opening the python3 console:
`sudo python3`
Then entering the following to light up 300 LEDs at once (if you are using more, just replace the number below):

```
import board
import neopixel
pixels = neopixel.NeoPixel(board.D18, 300)
pixels.fill((0,255,0))
```
This will light up all LEDs in red, if you see a bad one, mark it with some tape so you don’t use it for an airport.
Repeat the step for green:
`pixels.fill((255,0,0))`

Blue:
`pixels.fill((0,0,255))`

Purple:
`pixels.fill((0,125,125))`

Once you’re done you can turn off the lights by typing
`pixels.deinit()`

and then pressing CTRL+Z to exit the python console.

If the issue around the intensity also happens in this case, then it is a general problem between the Raspberry Pi and the Wiring, not so much to do with the script.

prueker, 2024-04-13:
Closing stale issue.

closed 2024-04-13 by prueker