prueker/METARMap #47: Changes to the API dataserver effective today

cbronner35, 2023-10-16:
Look like they finally did the changeover on aviationweather.gov and the dataserver has changed addresses. Which means the maps won't work. I was messing with it and trying to adapt the new dataserver to the code but we're medivac so of course I get a call right in the middle of it. If someone has already adapted the code please message me the new dataserver url string and save me a couple hours trying to figure it out lol

echo-alfa, 2023-10-16:
Looks like the new location is:

https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&hoursBeforeNow=4&format=xml&mostRecent=true&dataSource=metars&stationString=EGSS

cbronner35, 2023-10-16:
Yeah I've generated using the API tool but don't know how to adapt it to the metar.py. It looks like you just used that tool as well using a single station identifier. This won't help to convert the old dataserver to the new one and make the code work with it. Unless I'm missing something

echo-alfa, 2023-10-16:
The URL I’ve posted above is a direct replacement for the old url. Not had a chance to test it yet but you should be able to drop that into the metar.py line 146.

cbronner35, 2023-10-16:
It can't be a direct replacement especially if referencing EGSS on your stationstring. Unless your map only has one point which is London

echo-alfa, 2023-10-16:
OLD
url = "https://www.aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=5&mostRecentForEachStation=true&stationString=" + ",".join([item for item in airports if item != "NULL"])

NEW
url = "https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&hoursBeforeNow=5&format=xml&mostRecent=true&dataSource=metars&stationString=" + ",".join([item for item in airports if item != "NULL"])

cbronner35, 2023-10-16:
Sweet thanks! I have something similar but we got another call so I'll try
your when I get back to base and let you know!

mschultz21, 2023-10-16:
Thanks for finding this new URL! I was wondering why my map wouldn't work today.

Something I noticed with the new URL though, even if I put in a list of airports after stationString, it's only returning 1 airport at a time. Did you also encounter this error?

echo-alfa, 2023-10-16:
A few adjustments required.

`url = "https://aviationweather.gov/cgi-bin/data/dataserver.php?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=5&mostRecentForEachStation=true&stationString=" + ",".join([item for item in airports if item != "NULL"])`

I missed out the "ForEachStation" so it was only returning one result. The above solves that. 

Visibility seems to be returned as "6+" now and the + throws up and error. Fixed this with the below:

`if metar.find('visibility_statute_mi') is not None:
            visibility = metar.find('visibility_statute_mi').text
            if '+' in visibility:
                visibility = visibility.replace('+', '')  # Removing '+' character
            vis = int(round(float(visibility)))`

mschultz21, 2023-10-16:
Using your logic to replace the '+' with '' is giving me a syntax error for some reason, specifically pointing to this part `visibility: visibility = visibility.replace('+', '')`

I'm not well versed in Python, specifically shorthand notations so I'm not sure what's going on here

echo-alfa, 2023-10-16:
[image]

Make sure you have your indentation is correct.

mschultz21, 2023-10-16:
Yep that would be my problem! Thank you for pointing out my stupidity.

echo-alfa, 2023-10-16:
Hey, not at all. Easily done!

cbronner35, 2023-10-16:
Nice echo-alfa! Worked like a charm! Appreciate you diving into this my crew and I appreciate it!

prueker, 2023-10-17:
Thanks for the quick discussion for the fix, I'll try to update the script by the weekend when I get a moment based on your suggested fix here.

prueker, 2023-10-18:
The issue was fixed with #48 thanks to @moisedo

closed 2023-10-18 by prueker