prueker/METARMap #54: METAR Map shows all LEDs as blue occasionally

jburke26, 2023-12-18:
Occasionally I catch the metar map showing all blue LEDs, it appears to be hung.

If I manually initiate the refresh.sh it will fix the issue. It seems to happen more often if I lower the crontab file to refresh every 5 minutes vs 15-20 minutes.

I am using a Raspberry Pi Zero 2 W, has anyone experienced this?

jcl-rv12is, 2023-12-21:
Have had a couple maps running for many months, always set to 5 minute updates.  Never had that issue.  Using raspberry Pi Zero W.

bushflyin, 2024-03-01:
I have a similar issue. I don't think it's related to the crontab as I've commented that out for troubleshooting. I've noticed anytime the looplimit is set to anything other than 1, as it loops, it seems to occasionally lose track of what the LED was set to and changes. Watching the text scroll by it's continuously setting the airport to the same value, but what's reflected in the output, is not always reflected on the LED's. 

The cycling of the lights is in sync with the loop cycle. If I set it to 10 seconds, it's slower, but still occurs. I've tried pulling down the latest copy of metar.py, but still have the same issue. This is an initial install for me, so it's never quite worked right. It is accurate probably 80% of the time, but with 1 second set as the blink/fade interval, you can really see the issue frequently.

I've tested this with just one airport in the airports file and get this same issue. If I disable the display (and don't configure the wind or lightning animation), then there's no loop and the issue obviously goes away at that point.

Here's a short video (sorry about the background music) of what I'm experiencing: https://photos.app.goo.gl/CX72XsvqJTXNgG1Q7