prueker/METARMap #55: Pico W

emersonsc, 2024-01-05:
Have you tried this out with a Pico W?

I honestly think this project would be great with a Pico W as the computing power of the Zero is total overkill for this project. 

I'd love to give it a go, but I lack in the coding department.

cbronner35, 2024-01-05:
I might be wrong but I think it's a limitation of one of the libraries possibly the neopixel one. I believe it's got to do with the ram size

emersonsc, 2024-01-15:
> I might be wrong but I think it's a limitation of one of the libraries possibly the neopixel one. I believe it's got to do with the ram size

Nope. Plenty of room for the neopixel micropython library. In fact, I took sections of this code just to test 150 LEDs on a pic with random colors and it worked just fine. The road block im hitting at the moment is what command to pull the data from the website. The coding is slightly different all the way around for the LEDs as well as everything else so its going to take time to figure it out, but i REEALLY think this project could benefit from using a pico-w versus a Zero ONE because it will bring the cost down, and two, the zero is just way overkill for this project. Heck its likely possible to run it on an ESP32 board as well, but I have tons of Pico Ws I want to use for something and its small flat form factor is screaming to be used for this.

cbronner35, 2024-01-15:
Yeah wasn't sure about that. Well good luck maybe someone else can help you that's done it. The zeros only $15 I'd just do that since you lack the coding experience to rewrite the script. If you want to try to do it yourself you can always run the code through chatgpt and see if it can help you get an idea on what direction to take

emersonsc, 2024-01-15:
> Yeah wasn't sure about that. Well good luck maybe someone else can help you that's done it. The zeros only $15 I'd just do that since you lack the coding experience to rewrite the script. If you [...]

I've heard that used in the high school robotics-sphere to double check their code several times... no idea why i didnt think to do that here

cbronner35, 2024-01-15:
Yeah I use it when I'm stuck on a problem with a piece of code. It's not perfect at all and I wouldn't rely on it completely but it does give you different ideas on how to procede. Good luck hopefully you get it!