prueker/METARMap #56: Random changing LED Colors

emersonsc, 2024-01-07:
I've followed the directions several times, using bullseye and bookworm, how ever even using your code without touching any options, once the pi connects and get the METAR data, the screen does as it should, but the LEDs light up then change to random colors every 1 second.

cbronner35, 2024-01-08:
Do they change colors every 1 second or is more random on the timing? Just wondering if it's a grounding issue or some other issue. If your using a separate power supply did you attach an additional ground to the PIs ground? Sorry just trying to rule stuff out

emersonsc, 2024-01-08:
> Do they change colors every 1 second or is more random on the timing? Just wondering if it's a grounding issue or some other issue. If your using a separate power supply did you attach an additional ground to the PIs ground? Sorry just trying to rule stuff out

The Pi 3A+ and the string of LEDs (exact same ones linked from the write up) are powered by a meanwell 5v 10a PSU. A few times I got the LEDs to stay solid and had some flash (as they were supposed to because of the gusts) how ever the colors were wrong. Those that were supposed to be blue were green, some that were supposed to be blue were red. Nothing honestly made sense. I checked the strand with a Pico and found the strand was RGB not GRB and made the changes to the code and color calls but 9/10 times the screen does exactly what it's supposed to but the lights go into Christmas tree mode and cycle random colors and brightness. I tried a Pi Zero 2 W and this Pi 3A+ with the same results. I also tried bullseye and bookworm.

emersonsc, 2024-01-08:
Also, its every 1 second. Every once in a while there would be a maybe 3 second pause and then keep doing it. I've swapped wires, connectors, etc. The Red, green, and blue LEDs work correctly in all of the lights, again tested this via a Pico and Thonny.

Attached are my files.
[metar.zip](https://github.com/prueker/METARMap/files/13856084/metar.zip)

cbronner35, 2024-01-08:
That's an interesting problem I've had a similar problem on a hyperbian setup one time when I had a rpi and an led strip both connected directly to the power supply. What weirdly fixed that issue was I ran an additional ground from the led strip directly to the gpio gnd pin. Worth a shot if you haven't already. But sorry other than that I'm not sure maybe someone else has an idea. If I think of anything else I'll msg

emersonsc, 2024-01-15:
Was a loose ground. Its VERY important that if you are using (are really you should after about 20 LEDS) an external PSU for the LEDS, that you have a ground going to the RPi as well for the data line. Its also not a bad idea to put in a between a 400-500ohm resistor on the data line to clean up the signal. I also found amongst other guides that its not a bad idea to put a capacitor (several reviews for the LEDs the creator of this program link to suggested this) of 1000uF. This helps stabilize the voltage for the LEDs especially when brightness is high or you have lots of "white" LEDs going as they are power hungry.

closed 2024-01-15 by emersonsc

cbronner35, 2024-01-15:
Well I'm glad you found the solution!