prueker/METARMap #58: Wind Blinking kills updating

apldan, 2024-01-11:
I have cron running every 5 minutes, and I have the blink duration set for 300 seconds. It seems to blink correctly for the first 5 minute window, then freezes and stops updating. Any thoughts?

apldan, 2024-01-11:
Also, i hook my Pi up to a monitor while testing/changing things. is there a way to monitor what is being run? having a cmd window open only shows if i run the command by hand, not letting cron run it.

jcl-rv12is, 2024-01-12:
As the pi user, you could always run: tail -f /var/log/syslog.  That would only show you the command run every 5 minutes.  If you're looking for the detail output from refresh.sh, you'd have to edit the crontab entry to send standard output somewhere.  I would not recommend that.  **If sent to a file, it would get large quickly, so whatever you do, don't leave it set for long.**

*/5 6-22 * * * /home/pi/refresh.sh > /home/pi/output.log

You'd then "tail -f /home/pi/output.log". There may be a lag in the writes, so be aware of that.

If you wanted to have it sent to your display, type "tty" at a prompt, and use that device in place of a physical file, e.g.

*/5 6-22 * * * /home/pi/refresh.sh > /dev/pts/0

Be aware that you can't stop that screen output once started...you'd have to kill the script and then edit crontab to reverse it.  You'd have a tough time doing that with the constant output...you'd likely want to login remotely to do so.

In other words, be careful what you wish for.

jcl-rv12is, 2024-01-12:
> I have cron running every 5 minutes, and I have the blink duration set for 300 seconds. It seems to blink correctly for the first 5 minute window, then freezes and stops updating. Any thoughts?

Show us your crontab entry.

apldan, 2024-01-14:
Cron:

*/5 * * * * /home/pi/METARMap/refresh.sh

refresh.sh:

usr/bin/sudo pkill -F /home/pi/offpid.pid
/usr/bin/sudo pkill -F /home/pi/metarpid.pid
/usr/bin/sudo /usr/bin/python3 /home/pi/METARMap/metar.py & echo $! > /home/pi/METARmap/metarpid.pid

metar.py:

#!/usr/bin/env python3

import urllib.request
import xml.etree.ElementTree as ET
import board
import neopixel
import time
import datetime
try:
	import astral
except ImportError:
	astral = None
try:
	import displaymetar
except ImportError:
	displaymetar = None

# metar.py script iteration 1.5.1

# ---------------------------------------------------------------------------
# ------------START OF CONFIGURATION-----------------------------------------
# ---------------------------------------------------------------------------

# NeoPixel LED Configuration
LED_COUNT			= 50			# Number of LED pixels.
LED_PIN				= board.D18		# GPIO pin connected to the pixels (18 is PCM).
LED_BRIGHTNESS		= .5			# Float from 0.0 (min) to 1.0 (max)
LED_ORDER			= neopixel.GRB		# Strip type and colour ordering

COLOR_VFR			= (255,0,0)		# Green
COLOR_VFR_FADE		= (125,0,0)		# Green Fade for wind
COLOR_MVFR			= (0,0,255)		# Blue
COLOR_MVFR_FADE		= (0,0,125)		# Blue Fade for wind
COLOR_IFR			= (0,255,0)		# Red
COLOR_IFR_FADE		= (0,125,0)		# Red Fade for wind
COLOR_LIFR			= (0,125,125)		# Magenta
COLOR_LIFR_FADE		= (0,75,75)		# Magenta Fade for wind
COLOR_CLEAR			= (0,0,0)		# Clear
COLOR_LIGHTNING		= (255,255,255)		# White
COLOR_HIGH_WINDS 	= (255,255,0) 		# Yellow

# ----- Blink/Fade functionality for Wind and Lightning -----
# Do you want the METARMap to be static to just show flight conditions, or do you also want blinking/fading based on current wind conditions
ACTIVATE_WINDCONDITION_ANIMATION = True # Set this to False for Static or True for animated wind conditions
#Do you want the Map to Flash white for lightning in the area
ACTIVATE_LIGHTNING_ANIMATION = True		# Set this to False for Static or True for animated Lightning
# Fade instead of blink
FADE_INSTEAD_OF_BLINK	= True			# Set to False if you want blinking
# Blinking Windspeed Threshold
WIND_BLINK_THRESHOLD	= 15			# Knots of windspeed to blink/fade
HIGH_WINDS_THRESHOLD	= 25			# Knots of windspeed to trigger Yellow LED indicating very High Winds, set to -1 if you don't want to use this
ALWAYS_BLINK_FOR_GUSTS	= True		# Always animate for Gusts (regardless of speeds)
# Blinking Speed in seconds
BLINK_SPEED		= 1			# Float in seconds, e.g. 0.5 for half a second
# Total blinking time in seconds.
# For example set this to 300 to keep blinking for 5 minutes if you plan to run the script every 5 minutes to fetch the updated weather
BLINK_TOTALTIME_SECONDS	= 298

[... 207 lines cut ...]
		if ACTIVATE_WINDCONDITION_ANIMATION == True:
			pixels[i+ OFFSET_LEGEND_BY + 5] = COLOR_VFR if not windCycle else (COLOR_VFR_FADE if FADE_INSTEAD_OF_BLINK else COLOR_CLEAR)    # windy
			if HIGH_WINDS_THRESHOLD != -1:
				pixels[i + OFFSET_LEGEND_BY + 6] = COLOR_VFR if not windCycle else COLOR_HIGH_WINDS  # high winds

	# Update actual LEDs all at once
	pixels.show()
	
	# Rotate through airports METAR on external display
	if disp is not None:
		if displayTime <= DISPLAY_ROTATION_SPEED:
			displaymetar.outputMetar(disp, stationList[displayAirportCounter], conditionDict.get(stationList[displayAirportCounter], None))
			displayTime += BLINK_SPEED
		else:
			displayTime = 0.0
			displayAirportCounter = displayAirportCounter + 1 if displayAirportCounter < numAirports-1 else 0
			print("showing METAR Display for " + stationList[displayAirportCounter])

	# Switching between animation cycles
	time.sleep(BLINK_SPEED)
	windCycle = False if windCycle else True
	looplimit -= 1

print()
print("Done")

jcl-rv12is, 2024-01-14:
I can't guarantee this is your issue, but it looks like you're a victim of your decision to not put things in /home/pi, then not adjusting for that everywhere, and then using incorrect case in one place.

The refresh.sh script you posted is this:

usr/bin/sudo pkill -F /home/pi/offpid.pid
/usr/bin/sudo pkill -F /home/pi/metarpid.pid
/usr/bin/sudo /usr/bin/python3 /home/pi/METARMap/metar.py & echo $! > /home/pi/METARmap/metarpid.pid

1. Incorrect directories in the first two lines.
2. Incorrect case in the third line...METARmap should be METARMap

Try using:

usr/bin/sudo pkill -F /home/pi/METARMap/offpid.pid
/usr/bin/sudo pkill -F /home/pi/METARMap/metarpid.pid
/usr/bin/sudo /usr/bin/python3 /home/pi/METARMap/metar.py & echo $! > /home/pi/METARMap/metarpid.pid

While you're at it, you should look at your lightsoff.sh script.

apldan, 2024-01-14:
I think you nailed it, seems to be working great now :)

P.S. - may want to add something to the read me/Pi setup instructions about modifying /boot/config.txt to uncomment dtparam=spi=on and change dtparam=audio=on to off. Couple of bumps i had to research to get it to work on my Pi 3B+.

Thanks again for your help!

closed 2024-01-14 by apldan

jcl-rv12is, 2024-01-14:
That’s interesting.  I’ve never had to change anything for a 3B+.  Have two that I built for friends over a year ago that are running fine.

 

Have also tested again today a Pi 2 model B V1.1 (with an Edimax WiFi adapter) bought way back in 2015 that works fine.  I just moved the SD card built on a Pi Zero W to the Pi 2.  All functions, including the small OLED dislay, work fine.  This particular SD card was just built yesterday with 2023-12-11-raspios-bookworm-armhf-lite.img.xz.

 

As an aside, it really becomes a challenge to work with the Pi.  With every OS release, or even just the passage of time, libraries get updated, things get deprecated, and I spend an inordinate amount of time getting things right.  Sure enough, even moving from 2023-10-10-raspios-bookworm-armhf-lite.img.xz to 2023-12-11-raspios-bookworm-armhf-lite.img.xz was challenging.  That, and issues with SD card corruption, make it a pain to help friends out.

 

FWIW, going forward, for my own usage I will likely keep using this software, but for any maps for friends, I’m using a commercial microcontroller based solution.  I’ve also got a fork of the WKHarmon microcontroller code that adds significant functionality.

proppilot, 2024-01-14:
Once it is working it is working and does not need any upgrading.  Mine worked for three years until the METAR URL was updated and then I needed to make that change but I never update any libraries etc in the PI

jcl-rv12is, 2024-01-14:
True, and I did the same thing on several friends’ maps, but building a new one from scratch today is a different story.  I keep a Word doc updated regularly to cover building something today, at

 

https://github.com/jcl-rv12is/METARMap