prueker/METARMap #59: Airport List issue/Aviation weather issue?

hagendaus77, 2024-01-19:
How do I verify that there isn't an issue with the website where the data is being pulled from.  my map stopped working on 1/16/24.  Find attached the result of your airport list (which works) and when I add my airport list and run.  The first error seems to point to the website.  I will attach my airport list to issue 
[image]
[image]
[user airport and pics.zip](https://github.com/prueker/METARMap/files/13983164/user.airport.and.pics.zip)
pi@raspberrypi:~ $ sudo python3 metar.py
Running metar.py at 18/01/2024 19:17
Wind animation:False
Lightning animation:False
Daytime Dimming:False
External Display:False
Rotating through all airports on LED display
https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&stationString=KCMX,KIMT,KMNM,KSUE,KOCQ,KGRB,KMTW,KSBM,KETB,KUES,KMKE,KENW,KUGN,KORD,KRPI,KRFD,KJUL,KRYV,KMSN,KUNU,KFLD,KOSH,KATW,KCLI,KEZS,KAIG,KRHI,KLNL,KAUW,KCWA,KSTE,KISW,KYS0,KDLL,KPUB,KDBQ,KPDC,KLSE,KCMY,KPBH,KIWD,KASX,KDYT,KOLG,KRCO,KEAU,KMSP,KGRB,KATW,KOSH&hoursBeforeNow=5&format=xml&mostRecent=true&mostRecentForEachStation=constraint
Traceback (most recent call last):
  File "/home/pi/metar.py", line 156, in <module>
    content = urllib.request.urlopen(req).read()
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 523, in open
    response = meth(req, response)
  File "/usr/lib/python3.9/urllib/request.py", line 632, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.9/urllib/request.py", line 561, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 641, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 500: Internal Server Error

jcl-rv12is, 2024-01-19:
Just copy the URL in your post into a browser…it pulls all the data for all your airports just fine.  I copied your airports file into my system and it ran just fine.

 

pic.of.github.airport.list.png (view on web) <https://github.com/prueker/METARMap/assets/157080078/8d3f60e6-e2f8-4867-871e-12a4f79f8e9d> 
pic.of.my.airport.list.png (view on web) <https://github.com/prueker/METARMap/assets/157080078/9fda6f3c-7718-4a78-a106-01d5ee7385f4> 
https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve <https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&stationString=KCMX,KIMT,KMNM,KSUE,KOCQ,KGRB,KMTW,KSBM,KETB,KUES,KMKE,KENW,KUGN,KORD,KRPI,KRFD,KJUL,KRYV,KMSN,KUNU,KFLD,KOSH,KATW,KCLI,KEZS,KAIG,KRHI,KLNL,KAUW,KCWA,KSTE,KISW,KYS0,KDLL,KPUB,KDBQ,KPDC,KLSE,KCMY,KPBH,KIWD,KASX,KDYT,KOLG,KRCO,KEAU,KMSP,KGRB,KATW,KOSH&hoursBeforeNow=5&format=xml&mostRecent=true&mostRecentForEachStation=constraint> &dataSource=metars&stationString=KCMX,KIMT,KMNM,KSUE,KOCQ,KGRB,KMTW,KSBM,KETB,KUES,KMKE,KENW,KUGN,KORD,KRPI,KRFD,KJUL,KRYV,KMSN,KUNU,KFLD,KOSH,KATW,KCLI,KEZS,KAIG,KRHI,KLNL,KAUW,KCWA,KSTE,KISW,KYS0,KDLL,KPUB,KDBQ,KPDC,KLSE,KCMY,KPBH,KIWD,KASX,KDYT,KOLG,KRCO,KEAU,KMSP,KGRB,KATW,KOSH&hoursBeforeNow=5&format=xml&mostRecent=true&mostRecentForEachStation=constraint

hagendaus77, 2024-01-19:
Well that is extremely frustrating.  when I copy the url for either my airport list that doesn't work or the Github example list of airports that does work I get this error page isn't working error below.  The only way I can get the program to run is using the github airport example file.  As soon as I edit the file and remove one station I get the error.    
[image]

jcl-rv12is, 2024-01-19:
Just click on the link below.  What happens?

https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve <https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&stationString=KCMX,KIMT,KMNM,KSUE,KOCQ,KGRB,KMTW,KSBM,KETB,KUES,KMKE,KENW,KUGN,KORD,KRPI,KRFD,KJUL,KRYV,KMSN,KUNU,KFLD,KOSH,KATW,KCLI,KEZS,KAIG,KRHI,KLNL,KAUW,KCWA,KSTE,KISW,KYS0,KDLL,KPUB,KDBQ,KPDC,KLSE,KCMY,KPBH,KIWD,KASX,KDYT,KOLG,KRCO,KEAU,KMSP,KGRB,KATW,KOSH&hoursBeforeNow=5&format=xml&mostRecent=true&mostRecentForEachStation=constraint>

hagendaus77, 2024-01-19:
That is flipping crazy.  it gave me the link below.  Went up and hit the links above and now they all work.  Loaded my airport file onto the Pi and now it works again.  Was the something going on with the linked website?  This is bonkers.  definition of insantity type stuff doing that same thing 50 times and then it suddenly works on 51?  completely baffled.  

I really appreciate you sticking with me on that.  

[image]

jcl-rv12is, 2024-01-19:
No interruptions of any kind, and I’m in a room with two different maps on a Pi and a microcontroller all day.

hagendaus77, 2024-01-19:
again.   Appreciate you sticking with me.  She is up and running again!

[image]

closed 2024-04-13 by prueker