prueker/METARMap #60: Type Error

broort22, 2024-04-13:
When running the metar script I'm getting the following response:

File "metar.py.txt", line 215, in <module>
    + str(lightning))
TypeError: can only concatenate str (not "NoneType") to str

I never changed anything with the file so not sure why this is coming up now.  Any help would be appreciated

spenadel, 2024-04-13:
Same problem here too.

traviscole, 2024-04-13:
A very quick workaround is to disable the console output by commenting out (#) the two print statements - 
Lines 207-215 and line 263

spenadel, 2024-04-13:
Does anyone know what happened on the backend to cause the problem?

HaveBlueXST, 2024-04-13:
Also having the same issue, but commenting out the console output doesn't seem to be working for me.

prueker, 2024-04-13:
Is this a new installation?
I can't seem to reproduce the problem - could you please in the print statement between 207-215 try to comment out the individual lines to see which exact line might be the problem to help narrow it down?

spenadel, 2024-04-13:
Mine started working without any intervention from me. Don't understand what happened.

broort22, 2024-04-13:
when i comment out 215 im getting a syntax error on line 216.  Sorry, im not very proficient with linux as of yet.

HaveBlueXST, 2024-04-13:
Not a new installation for me, but I just grabbed a fresh copy of metar.py in case there was some tweak I had done and forgotten about.  I just had to modify it to point to where I have my airports file in /home/metarmap.  Running it now gives me this:

metarmap@metarmap0:~ $ sudo python3 metar.py
[sudo] password for metarmap:
Running metar.py at 13/04/2024 17:46
Wind animation:False
Lightning animation:False
Daytime Dimming:False
External Display:False
Rotating through all airports on LED display
https://aviationweather.gov/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&stationString=KUGN,KENW,KRAC,KMKE,KMWC,KETB,KSBM,KMTW,KATW,KY50,KOSH,KFLD,KUNU,KRYV,KUES,K57C,KBUU,KJVL,KFEP,KEFT,KC29,KMSN,KDLL,KC35,K82C,KVOK,KCMY,KBCK,KLSE,KONA,KDEH,KY51,KPDC,KOVS,KLNR,KMRJ,KPVB,KDBQ,KMXO&hoursBeforeNow=5&format=xml&mostRecent=true&mostRecentForEachStation=constraint
K57C:VFR:210@11G17:10SM::20/2:29.87:False
Traceback (most recent call last):
  File "/home/metarmap/metar.py", line 207, in <module>
    print(stationId + ":"
TypeError: can only concatenate str (not "NoneType") to str

HaveBlueXST, 2024-04-13:
I think the issue may be from a specific airport.  My normal airports file is:
KUGN
KENW
KRAC
KMKE
KMWC
KETB
KSBM
KMTW
KATW
KY50
KOSH
KFLD
KUNU
KRYV
KUES
K57C
KBUU
KJVL
KFEP
KEFT
KC29
KMSN
KDLL
KC35
K82C
KVOK
KCMY
KBCK
KLSE
KONA
KDEH
KY51
KPDC
KOVS
KLNR
KMRJ
KPVB
KDBQ
KMXO

I just reduced the file to the first 5, and things are working!  Will expand the file and see which airport is breaking it.

prueker, 2024-04-13:
Ok, I found the issue, the airport **KFEP** is currently sending a empty flight category (instead of not sending that tag at all as it used to be the case if no flight category was currently being transmitted.
This caused the script to fail as it was expecting a value of the field was present in the data received from aviationweather.gov.
I just patched the issue with the change in 1002f562a72e1d188705e590aca8699cb59d11d2
Please apply the change made to metar.py on line 165 (adding the second part to the existing skipping logic to check if .text is also None):
`	if metar.find('flight_category') is None or metar.find('flight_category').text is None:
		print("Missing flight condition for " + stationId + ", skipping.")`

closed 2024-04-13 by prueker

HaveBlueXST, 2024-04-14:
Brilliant! Made the edit you suggested, and all is back to normal! You are a rockstar!!!