prueker/METARMap #66: neopixel install error

Birdheezy, 2024-05-30:
It appears that the neopixel install returns the error:

`error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
`

Krob99, 2024-07-14:
Same issue, did you ever find a fix?

Birdheezy, 2024-07-15:
They way bookworm handles the install of these kinds of things have changed. 
https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi?view=all

xcmtber14, 2024-11-04:
I tried reading through the changes for installing these but it's well over my head. Has anyone successfully installed and could update the files and readme?

Birdheezy, 2024-11-04:
> I tried reading through the changes for installing these but it's well over my head. Has anyone successfully installed and could update the files and readme?

https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi?view=all#create-the-venv-3157396

If you're on the newest version (bookworm) follow that link and head down to the section "Create the venv" Follow the instructions to create a virtual environment and activate it. Now you can install the neopixel library in the VENV using
`pip3 install adafruit-circuitpython-neopixel`
you can now deactivate the VENV with 'deactivate'.
You can now either activate the VENV  when you want to run the metar script or you can use
`sudo /home/pi/metar/bin/python3 metar.py`
where 'metar' is the name of your VENV.

jcl-rv12is, 2024-12-16:
You could consider using my Word document, which has been updated for this and a variety of software install/config challenges.

https://github.com/jcl-rv12is/METARMap

Tysseract, 2025-08-06:
I know I'm late to the party but the instructions say `You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.`
I did that and it works just fine...don't see how this little code could break the OS and if it does I can always reflash it.